MicrosoftDocs/windows-driver-docs #1211: off-screen-memory link is wrong

hsupu, 2019-01-09:
target https://msdn.microsoft.com/library/windows/hardware/ff556318#wdkgloss-off-screen-memory is redirect to https://docs.microsoft.com/en-us/windows-hardware/drivers/#wdkgloss-off-screen-memory, which is no different from https://docs.microsoft.com/en-us/windows-hardware/drivers/ .

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 74e64446-8a87-626a-9150-8d37e89d5869
* Version Independent ID: 30ce9cc9-1e7b-203b-b8d6-f53a173dde1c
* Content: [Creating Device-Dependent Bitmaps - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/creating-device-dependent-bitmaps)
* Content Source: [windows-driver-docs-pr/display/creating-device-dependent-bitmaps.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/creating-device-dependent-bitmaps.md)
* Product: **windows-hardware**
* GitHub Login: @mimisasouvanh
* Microsoft Alias: **misaso**

mimisasouvanh, 2019-01-09:
@hsupu I am looking into this now.

mimisasouvanh, 2019-01-12:
@hsupu Thank you for bring this to our attention. The link should be fixed to go to the correct description of off-screen memory. You'll see the changes by Monday.

closed 2019-01-12 by mimisasouvanh