MicrosoftDocs/windows-driver-docs #1212: Missing parameter 1 value

jakeoshins, 2019-01-09:
MEMORY_MANAGEMENT (1a)
    # Any other values for parameter 1 must be individually examined.
Arguments:
Arg1: 000000000006194a, The subtype of the bugcheck.
Arg2: 0000000000000000
Arg3: 0000000000000200
Arg4: 0000000000000001

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-01-17:
@jakeoshins - Thanks for this issue. Because we have not documented information on this parameter, I would need to check with the dev team to be able to do that. We use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes. 

Right now I'm working on the new stop codes that we have not documented at all.

Thank you for your interest in improving our stop code docs.

DOMARS, 2019-03-18:
@jakeoshins - I'm checking with the dev team to see if we can document the 6194A parameter 1.

DOMARS, 2019-06-20:
@jakeoshins - I added this description for to a month or so back, but missed closing this issue.

*0x6194A - An error occurred when decrementing the reference counts on I/O space physical pages which are being unmapped. An entry which is not currently referenced is being dereferenced.  Parameter 2 and 3 describe the caller’s I/O space range being unmapped, and parameter 4 is the I/O space physical page which is expected to be referenced but is not.*

Thanks for the interest in our docs and this issue.

closed 2019-06-20 by DOMARS