MicrosoftDocs/windows-driver-docs #1213: Tightly coupling multiple INF files

CKroid, 2019-01-11:
We have a driver package that that tightly couples the extension driver with component driver (using CopyInf), however, when we tried to create a shipping label for WU, it doesn't allow us to target the hardware device id listed in Extension INF, but only allow us to target the software component id. At such, the extension driver would be able to install, thus unable to install the software component.
How should Tightly coupling driver package be setup for WU?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5c233326-2cb5-a042-aa1c-7b1f11080250
* Version Independent ID: 2edce1a7-c0c2-116f-bbf7-4d46ae02d9f5
* Content: [Universal Driver Scenarios - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/universal-driver-scenarios)
* Content Source: [windows-driver-docs-pr/develop/universal-driver-scenarios.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/universal-driver-scenarios.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

CKroid, 2019-01-12:
If I understand you correctly, the Tightly coupled Extension and Component scenario described above is not allowed.
My next question will be is it possible to limit the version of software components depending on the version of extension installed? Something similar to how UWP application version limited by driver version found in the system.

tedhudek, 2019-01-14:
Sorry @CKroid this is not possible.  Why is such a tight coupling needed?  They are already separate driver installations.

CKroid, 2019-01-14:
We have a software component driver which we hope to share between OEM and Retail customers. However, some OEMs would want to limit the distributions of this software component driver by specific CHIDs. Once we specify any CHIDs in the software component WU Shipping label, Retail customers will no longer be targetted since they will not have specific CHID.
However, we will be able to easily create 2 shipping labels in extension driver, one for OEM HWIds, one for retail HWIds. If the component driver can be tightly coupled with the extension driver, OEM will be able to get the WU update of the extension + component according to the CHID specified, and retail customers will continue to receive WU for extension + component without any restrictions.

tedhudek, 2019-01-14:
Hi, you can do the same kind of publishing that you're doing with the extension to the software component.

In total, you would have 4 submissions – 2 for the extension and 2 for the component.

Each would have one that is CHID targeted and one that is not.  The tight coupling itself shouldn’t be required.

closed 2019-01-14 by tedhudek

CKroid, 2019-01-15:
Oh? That is good information!
Considering the following scenario:
Extension v1 and Component v1 targeted to CHID1 was posted to WU.
Subsequently, I posted a newer Extension v2 and Component v2 with no CHID target.
System with CHID1 will still only see Extension v1 and Component v1 despite there is a higher version available in WU but not targetting CHID1?

tedhudek, 2019-01-16:
Yes, that's correct.