MicrosoftDocs/windows-driver-docs #1219: WU submission approval is taking longer than normal.

BretSagan, 2019-01-15:
We have an optional WU submission pending for over 1 month.
Normally it takes less than one hour for optional WU, but this one is pending without any error message.
Since it's Intel onboard graphics WU, we've make sure we only published 4 part ID inf files.
Is there any way we can see error messages or speed up submission procedure?

https://partner.microsoft.com/en-us/dashboard/hardware/driver/14104289654184567/submission/1152921504627873217/ShippingLabel/1152921504607355672

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 079ebacf-2baa-b380-dd28-ade6141a7f02
* Version Independent ID: eb3ac6de-78cd-8ca3-a871-e06062dc0fc3
* Content: [Hardware dashboard FAQ - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-dashboard-faq#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/hardware-dashboard-faq.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/hardware-dashboard-faq.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-03-06:
@BretSagan, sorry I didn't see this issue logged.  
@SillyKeith 
I cannot see if this submission has been resolved. Can you please look?
Thanks,
Eliot

BretSagan, 2019-03-07:
@EliotSeattle @SillyKeith This issue has been resolved, thank you.

closed 2019-03-07 by EliotSeattle