MicrosoftDocs/windows-driver-docs #1225: Where has Windows Hardware Developer Center Dashboard moved to?

NetLib, 2019-01-15:
The link above now takes me to the Partner Dashboard. Where do I find all my driver packages previously signed by Microsoft for Windows 10?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 138a9602-43a6-1b8a-7159-dc3cd7e62f5a
* Version Independent ID: dbf5e2e6-51e1-a43b-becb-3919dc51cde3
* Content: [Windows 10 in S mode Driver Requirements - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/windows10sdriverrequirements)
* Content Source: [windows-driver-docs-pr/install/Windows10SDriverRequirements.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/Windows10SDriverRequirements.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-01-15:
Hi @NetLib are you looking for https://developer.microsoft.com/en-us/windows/hardware/dashboard-sign-in?

You can find docs for the dashboard at https://docs.microsoft.com/windows-hardware/drivers/dashboard/.

NetLib, 2019-01-16:
Well it took the better part of a day but I finally got back in. I re-registered as a Partner, and then registered as a Hardware developer and then finally was able to reach the signing portal again. Unfortunately all my old packages are gone, but at least I am able to sign new packages.

closed 2019-01-16 by NetLib