MicrosoftDocs/windows-driver-docs #1233: The description of Network Layer IP Fragment confused

GreatFireWall, 2019-01-17:
Although the document is very detailed, but I still feel confused.

Accroding to [wfp-layer-requirements-and-restrictions](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/wfp-layer-requirements-and-restrictions):
<pre>
...
<b>Network Layer</b>

IP packet fragments, which are indicated only for incoming paths, 
<b>are indicated three times at this layer--first as an IP packet, 
again as an IP fragment, and a third time as part of a reassembled IP packet.</b>[1] 
WFP sets the FWP_CONDITION_FLAG_IS_FRAGMENT flag when it indicates fragments 
to network layer callouts.

When adding filtering conditions, FWP_MATCH_FLAGS_NONE_SET can be used together 
with the FWP_CONDITION_FLAG_IS_FRAGMENT flag to avoid the second indication. 
<b>If the callout has to inspect only full packets (those that have not been fragmented and reassembled),
it has to parse the IP header to avoid processing fragments that are indicated as IP packets.</b>[2]
Alternatively, the callout can inspect packets at the transport layer.
...
</pre>

--------------

**About mark [1]:** 

The below `MF` means `More Fragments Flag`, `FO` means `Fragment Offset` those definition in IP header.
p1 p2 pn means pakcet1 packet2 packetn.

Now, assuming there're 4 fragmented packets incoming at INBOUND_IPPACKET:
`local <- p1[MF=1,FO=0] <- p2[MF=1,FO=185] <- p3[MF=1,FO=310] <- p4[MF=0,FO=495] <- remote`

_My question is:_ 
Accroding to mark[1], how does WFP represent 4 packets as three times ? 
In other words, will my InboundIPPacketClassifyFn called 3 times or 4 times ? 
If 3 times, where does the lost packet go ?

**About mark [2]:** 

_My question is:_ 
Accroding to `third time as part of a reassembled IP packet` and mark[2], if I want to receive only full packet (reassembled), could I simply do `if (packet->MF == 0)` for determine it is a reassembled packet ?

#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3977d9ce-9be4-06fd-0ed6-dc30d323eac6
* Version Independent ID: 0c2f16cc-05b2-2873-7060-c3872542baf4
* Content: [WFP Layer Requirements and Restrictions - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/wfp-layer-requirements-and-restrictions)
* Content Source: [windows-driver-docs-pr/network/wfp-layer-requirements-and-restrictions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/wfp-layer-requirements-and-restrictions.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-01-18:
Hi @GreatFireWall, thanks for the questions. First, sorry for closing/reopening the issue several times. I was ironing out a bug in our automatic assign bot and had to use this issue to test the fix. :)

As for your questions, I don't have the final answer to either of them, sorry. Since this isn't really a troubleshooting question and is aimed at enhancing/clarifying the docs, I'll go ahead and email some of the WFP folks to see if we can get more information here and improve the doc. Hang tight and thanks for your patience.

duncanmacmichael, 2019-01-23:
Hi @GreatFireWall, I've gotten a response for you that I hope clears things up.

1. In your example, there would be 9 indications.
    1. One indication for each "original" IP packet (4 classifications, or calls to the driver's classify function)
    2. One indication for each "original fragment" (4 classifications)
    3. One indication for the final reassembled IP packet (1 classification)
2. To determine if a packet is reassembled, you can do these steps:
    1. Check the More Fragments flag and/or see if the Fragment Offset field is not 0
    2. Write a filter that allows all classifications where **FWP_CONDITION_FLAG_IS_FRAGMENT** is set.
    3. Do whatever processing you need on the reassembled packet.

I've added this information to the doc, and that should be live later tonight or tomorrow morning (Pacific time). Thanks again for the questions, since I think they helped add clarity to the doc.

closed 2019-01-23 by duncanmacmichael

GreatFireWall, 2019-01-23:
1. One indication for each "original" IP packet (FWP_CONDITION_FLAG_IS_FRAGMENT **not set**)
2. One indication for each "original fragment" (FWP_CONDITION_FLAG_IS_FRAGMENT **set**)

Is it right ?

duncanmacmichael, 2019-01-23:
That's correct as far as I understand it.

GreatFireWall, 2019-01-24:
understood, thanks your guide :)