MicrosoftDocs/windows-driver-docs #1260: "Create a new hardware logo submission" link broken

GabeSJones, 2019-01-23:
That link goes to https://msdn.microsoft.com/library/windows/hardware/br230808.aspx, which ends up redirecting to the main dashboard page at https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7135005c-e7aa-19a2-7cfe-5003696da66a
* Version Independent ID: dba3259f-cb98-c2ab-9c56-860a731310d9
* Content: [Hardware submissions - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-certification-submissions)
* Content Source: [windows-driver-docs-pr/dashboard/hardware-certification-submissions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/hardware-certification-submissions.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-01-23:
Hi @GabeSJones.
That's a good one. I think there may be a couple of little edits required, including having the correct redirect. Let me do a little research, and I'll get back to you.

EliotSeattle, 2019-01-24:
Added the right link and the right description. Thank you!

closed 2019-01-24 by EliotSeattle