MicrosoftDocs/windows-driver-docs #1261: Several Broken Links to "Standard Driver Routines" Page

rbroker, 2019-01-23:
Hi,

Several of the docs refer to a "Standard Driver Routines" page (https://msdn.microsoft.com/library/windows/hardware/ff563842) which used to exist in the old MSDN docs, but doesn't seem to have an equivalent in the new docs (it just seems to redirect the [index page](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/index)).

The following docs appear to make use of this link:

- kernel/always-preemptible-and-always-interruptible.md
- kernel/driverentry-s-required-responsibilities.md
- kernel/introduction-to-plug-and-play.md
- kernel/introduction-to-standard-driver-routines.md
- kernel/overview-of-windows-components.md
- kernel/types-of-windows-drivers.md
- kernel/using-ntstatus-values.md
- storage/introduction-to-storage-class-drivers.md
- storage/storage-class-driver-s-standard-driver-routines.md

There seems to be a version of it still online [here](https://msdn.microsoft.com/de-de/vstudio/ff563842(v=vs.120)). I was initially thinking of making a pull request to add a new file containing that content and linking to the appropriate parts of the current documentation, but looking at it further I'd say anywhere that currently links to "[ff563842](https://msdn.microsoft.com/library/windows/hardware/ff563842)" is probably better off linking to [Introduction to Standard Driver Routines](https://docs.microsoft.com/windows-hardware/drivers/kernel/introduction-to-standard-driver-routines) instead (and then removing the recursive link from that page to itself) because that page contains a bunch more useful information (possibly improved by turning the entries in the leftmost column into hyperlinks (e.g. linking the "DriverEntry" cell to https://msdn.microsoft.com/library/windows/hardware/ff544113).

However, that seems like more of a judgement call on the MS side, so I'm opening it as an issue instead. Thanks for your time!

\- Richard

EliotSeattle, 2019-01-24:
Thanks, @rbroker, for the excellently compiled Issue. Your suggested use of  Introduction to Standard Driver Routines with the addition of the links sounds perfect. We're a little backed up with work right now, but either @tedhudek or I will take this on.

tedhudek, 2019-02-14:
Thanks @rbroker for the thoughtful suggestions.  I've just updated the .md source in b16e16350, this will refresh on the live site overnight tonight PST

closed 2019-02-14 by tedhudek

rbroker, 2019-02-14:
Looks, great, thank you!