MicrosoftDocs/windows-driver-docs #1262: Dead link

JGondek, 2019-01-24:
"Get (WinDbg) as part of Windows 8.1 SDK"
https://www.microsoft.com/click/services/Redirect2.ashx?CR_EAC=300135395

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3f5d6d55-cb35-6b9e-f8ca-e75f70826dc3
* Version Independent ID: 1d913516-b1e0-e872-c337-e30ed5dffce8
* Content: [Previous WDK versions and other downloads - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads)
* Content Source: [windows-driver-docs-pr/other-wdk-downloads.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/other-wdk-downloads.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2019-01-24:
Thanks, @JGondek! 
I'll try to track down the issue ASAP.

EliotSeattle, 2019-01-24:
Fixed in source. Use https://go.microsoft.com/fwlink/p/?LinkId=323507. Will be live later today.
Thanks, @JGondek.

closed 2019-01-24 by EliotSeattle