MicrosoftDocs/windows-driver-docs #1263: Declaritive => Declarative

GabeSJones, 2019-01-24:
Declaritive =&gt; Declarative

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7135005c-e7aa-19a2-7cfe-5003696da66a
* Version Independent ID: dba3259f-cb98-c2ab-9c56-860a731310d9
* Content: [Hardware submissions - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-certification-submissions)
* Content Source: [windows-driver-docs-pr/dashboard/hardware-certification-submissions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/hardware-certification-submissions.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

GabeSJones, 2019-01-24:
Also, seperated => separated

EliotSeattle, 2019-01-24:
Ooops. Fixed in source. Building to staging server now. Thanks, @GabeSJones

EliotSeattle, 2019-01-24:
All foxed. Thanks, @GabeSJones

EliotSeattle, 2019-01-24:
fixed. :)

closed 2019-01-24 by EliotSeattle