MicrosoftDocs/windows-driver-docs #1278: Values for parameter 1

LuigiBruno, 2019-01-25:
Are the possible values for the parameter 1 the same ones listed at https://msdn.microsoft.com/library/cc704588.aspx?
If not, is there a list of those values?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 76528be9-25fa-14da-0f12-98e650278efa
* Version Independent ID: 22ae5b42-6beb-81b5-7652-1085a3bb92e6
* Content: [Bug Check 0x6B PROCESS1_INITIALIZATION_FAILED - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x6b--process1-initialization-failed)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x6b--process1-initialization-failed.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x6b--process1-initialization-failed.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-02-01:
@LuigiBruno - Thanks for this issue. As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.
Right now I'm working on the new stop codes that we have not documented at all.
Thank you for your interest in improving our stop code docs.

DOMARS, 2020-12-22:
@LuigiBruno - Yes it does appear that this param1 is a standard NT STATUS code listed in the link you provided. As this failure is normally either hardware or something in Windows, not sure how helpful this code will be. As the topic mentions, I would recommend troubleshooting the hard drive system. For example if the vendor provides utilities.  You may not be able to boot, so might need to boot from bootable media, such as a thumb drive first. As this is a less frequent bug code, and there are a few new bug codes that need to be documented, I think I will leave this one as is for now.  Thank you for your interest in improving our stop code docs.

closed 2020-12-22 by DOMARS

LuigiBruno, 2021-01-18:
> @LuigiBruno - Yes it does appear that this param1 is a standard NT STATUS code listed in the link you provided. As this failure is normally either hardware or something in Windows, not sure how [...]

OK, @DOMARS. Thank you.