MicrosoftDocs/windows-driver-docs #1286: Parameters description

LuigiBruno, 2019-01-26:
This bugcheck has four parameters: which is the meaning of each of them?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f07c0a31-ec8e-3a19-0ad8-07b38665c008
* Version Independent ID: f8e51119-8145-c30d-883a-209b127d59b5
* Content: [Bug Check 0x80 NMI_HARDWARE_FAILURE - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x80--nmi-hardware-failure)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x80--nmi-hardware-failure.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x80--nmi-hardware-failure.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-02-01:
@LuigiBruno - Thanks for this issue. As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.
Right now I'm working on the new stop codes that we have not documented at all.
Thank you for your interest in improving our stop code docs.

DOMARS, 2019-09-11:
@LuigiBruno - I have opened an internal work item to track this work, so closing this issue for now.

As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes.

Thank you for your interest in improving our stop code docs.

closed 2019-09-11 by DOMARS