MicrosoftDocs/windows-driver-docs #1291: Binscope is AWOL

snowkoan, 2019-02-01:
The section entitled "Check code with BinScope Binary Analyzer" needs a revamp. In particular, I can't find a working link to download the tool. The following links are broken:

https://www.microsoft.com/download/details.aspx?id=44995
https://blogs.microsoft.com/microsoftsecure/2014/11/20/new-binscope-released/

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6298d0a1-1f6b-6e00-5445-dfec501a4430
* Version Independent ID: 42f902a4-b1df-b863-7486-de26c54a205f
* Content: [Driver security checklist - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/driversecurity/driver-security-checklist#binscope)
* Content Source: [windows-driver-docs-pr/driversecurity/driver-security-checklist.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/driversecurity/driver-security-checklist.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-02-01:
@snowkoan - Thanks for pointing this out. It does look like the blog is down or being moved. The download link is working for me. It could have been down for a bit, but seems to be working for me this morning.

snowkoan, 2019-02-01:
Thanks. That's bizarre. In Chrome, I get the following for the download link:
"We are sorry, the page you requested cannot be found."

In Edge, it works fine.  I've never seen that before. Same system.

rodwiddowson, 2019-02-02:
@snowkoan I'm seeing the download link OK in Chrome/Windows (+Chrome/IoS) from Scotland.  Some local mirroring issue?

PS: hi!

DOMARS, 2019-02-06:
@snowkoan - The link to the blog now goes to it's new location: https://cloudblogs.microsoft.com/microsoftsecure/
And there is a redirect in place to forward existing links. Thanks for letting us know there was an issue.

closed 2019-02-06 by DOMARS