MicrosoftDocs/windows-driver-docs #1294: Additional content

LuigiBruno, 2019-02-01:
I wrote an article about bugcheck analysis: it is published at https://social.technet.microsoft.com/wiki/contents/articles/6302.windows-bugcheck-analysis.aspx.
Do you think its content could be useful as an addition to the information provided in this page?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8650617e-670b-1ced-4404-d6011ea4b3b1
* Version Independent ID: 68feb97d-078c-452c-ffeb-473008362382
* Content: [Interpreting a Bug Check Code - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/interpreting-a-bug-check-code)
* Content Source: [windows-driver-docs-pr/devtest/interpreting-a-bug-check-code.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/interpreting-a-bug-check-code.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-02-05:
Sorry, @LuigiBruno, but that content is very dated. Thanks for the offer, though.
Eliot

closed 2019-02-05 by EliotSeattle

LuigiBruno, 2019-02-05:
Thanks @EliotSeattle.
I'm updating that article to provide more examples, but it describes the basic information and steps to understand a bug check error, analyze the memory dump file and understand the root cause: I'm constantly applying the steps that I've described there every time I troubleshoot a bug check error, whether in TechNet Forums or not.