MicrosoftDocs/windows-driver-docs #1298: Duplicate page?

LuigiBruno, 2019-02-03:
Which is the difference between this bug check and the 0xDEADDEAD one?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 24037ce4-4e17-5b20-38a4-f786066e0813
* Version Independent ID: b3223005-3b41-f5f9-cf4f-55706c21209f
* Content: [Bug Check 0xE2 MANUALLY_INITIATED_CRASH - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xe2--manually-initiated-crash)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xe2--manually-initiated-crash.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xe2--manually-initiated-crash.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-02-06:
@LuigiBruno - Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.
You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.
•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

LuigiBruno, 2019-03-06:
Thank you @DOMARS. I'll do that.

DOMARS, 2019-03-11:
@LuigiBruno - Reopening this issue to see if I can gather any information from the dev team on why there are two versions of  MANUALLY_INITIATED_CRASHx.

DOMARS, 2019-03-11:
@LuigiBruno - Two ways to manually initiate 0xE2 crash dumps according to these topics. I have not tested this. 

[Forcing a System Crash from the Keyboard](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/forcing-a-system-crash-from-the-keyboard)
[Forcing a System Crash from the Debugger](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/forcing-a-system-crash-from-the-debugger)

DOMARS, 2020-02-25:
@LuigiBruno - I worked with the dev team to update 0xDEADDEAD. The 0xDEADDEAD code is not invoked with the .crash command or using the keyboard, so this is a different stop code.

closed 2020-02-25 by DOMARS