MicrosoftDocs/windows-driver-docs #1299: Duplicate page?

LuigiBruno, 2019-02-03:
Which is the difference between this bug check and the 0x000000E2 one?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 09f4b152-0641-b32a-7189-68b8ad0110d7
* Version Independent ID: c5a601c5-4dff-f5ad-8c0e-a4bd7f18d179
* Content: [Bug Check 0xDEADDEAD MANUALLY_INITIATED_CRASH1 - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xdeaddead--manually-initiated-crash1)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xdeaddead--manually-initiated-crash1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xdeaddead--manually-initiated-crash1.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-02-06:
@LuigiBruno - Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.
You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.
•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

closed 2019-02-06 by DOMARS

LuigiBruno, 2019-03-06:
Thank you, @DOMARS. I'll do that.