MicrosoftDocs/windows-driver-docs #1300: Assemble command a misbehaviour

blabberer, 2019-02-04:
the assemble a command misbehaves when passing symbols instead of numeric 

it appears it has been so from at least build 14951 to the latest at the time of posting 17763

0:000> ? calc+0x1000
Evaluate expression: 13373440 = 00cc1000
0:000> a .
771f05a6 mov eax , calc+0x1000
771f05ab
0:000> u . l1
ntdll!LdrpDoDebuggerBreak+0x2c:
771f05a6 a10010cc00      mov     eax,dword ptr [calc!_imp__SHGetSpecialFolderPathW (00cc1000)]
0:000> a .
771f05a6 mov eax , cc1000
771f05ab
0:000> u . l1
ntdll!LdrpDoDebuggerBreak+0x2c:
771f05a6 b80010cc00      mov     eax,offset calc!_imp__SHGetSpecialFolderPathW (00cc1000)
0:000>

as can be seem in the first trial a **symbol + offset** is assembled as a de-reference  which is not intended while passing an actual numerical address  assembles correctly 

also this command doesn't seem to follow the  module name syntax rules which states add an exclamation mark to differentiate between ambiguous module names (viz cdb) which could be interpreted as numeric  

```
0:000> a
771f05a6 mov eax , calc+0x1000
771f05ab mov eax , !calc+0x1000
        ^ Syntax error in 'mov eax , !calc+0x1000'
771f05ab mov eax , calc!+0x1000
        ^ Couldn't resolve 'mov eax , calc!+0x1000'
771f05ab mov eax , calc! + 0x1000
        ^ Couldn't resolve 'mov eax , calc! + 0x1000'
771f05ab mov eax , !calc! + 0x1000
        ^ Syntax error in 'mov eax , !calc! + 0x1000'
771f05ab mov eax , !calc + 0x1000
        ^ Syntax error in 'mov eax , !calc + 0x1000'
771f05ab mov eax , (!calc + 0x1000)
        ^ Syntax error in 'mov eax , (!calc + 0x1000)'
771f05ab mov eax , @@masm(!calc + 0x1000)
        ^ Bad register error in 'mov eax , @@masm(!calc + 0x1000)'
771f05ab mov eax , @@masm(calc + 0x1000)
        ^ Bad register error in 'mov eax , @@masm(calc + 0x1000)'
771f05ab mov eax , @@masm(calc! + 0x1000)
        ^ Bad register error in 'mov eax , @@masm(calc! + 0x1000)'
771f05ab mov eax , @@c++(calc! + 0x1000)
        ^ Bad register error in 'mov eax , @@c++(calc! + 0x1000)'
771f05ab mov eax , (calc + 0x1000)
771f05b0 u . l2
        ^ Bad opcode error in 'u . l2'
771f05b0
0:000> u . l2
ntdll!LdrpDoDebuggerBreak+0x2c:
771f05a6 a100105a00      mov     eax,dword ptr [calc!_imp__SHGetSpecialFolderPathW (005a1000)]
771f05ab a100105a00      mov     eax,dword ptr [calc!_imp__SHGetSpecialFolderPathW (005a1000)]
0:000>
```

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e234f22d-8c0b-d7ad-0a2f-722fb60388e9
* Version Independent ID: e3bec727-c202-4a53-bd79-2a87e7d936d6
* Content: [a (Assemble) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/a--assemble-)
* Content Source: [windows-driver-docs-pr/debugger/a--assemble-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/a--assemble-.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-02-21:
@blabberer - Thanks for reaching out to the Windows Driver Developer Documentation team. My assembly is pretty rusty and I think you are raising a number of issues here that I may not be tracking very well.  I have added someone with debugger team here - Andy for their awareness. It looks like your first sample output is from 32 bit Windows and your second sample is from 64 bit Windows? 

This old OSR thread indicates that there may have been issues with 64 bit and the assemble command.

https://community.osr.com/discussion/191038/assemble-command-a-64-bit-windbg

BTW, the OSR forums are a great place for digging into the finer points if the use of WinDbg.
[OSR's windbg List](https://community.osr.com/categories/windbg)

If you believe that there is a product issue, I would create a single issue for each, with the format of current behavior and expected behavior with sample output that can be reproduced. 

For product input follow this link to file product feedback using the Windows Feedback Hub: 
http://aka.ms/dexex 
The Feedback Hub allows customers to file and up vote product issues. The link automatically sets the product category to Debugging Tools for Windows.

If you believe there are issues with the documentation, you can attach an issue to a specific topic, or create a PR with a specific fix. 

Thanks again for reaching out to the Windows Driver Developer Documentation team.

blabberer, 2019-02-21:
@DOMARS 
1) no they are all in 32 bit 
2) I am Aware that Assemble command Does not support  64 bit mnemonics 
3) I am Aware of osr's windbg list

DOMARS, 2020-07-25:
@blabberer - Apologies on the delay here. I have not been able to gather any additional information on the documentation side, so I think the best path forward is to file product feedback for the Windows debugger dev team, using the http://aka.ms/dexex link.  I just tested the product input link and it is working for me. Do let me know if that link is not working for you. 

@aluhrs13 - FYI on incoming product input on the assemble command.

closed 2020-07-25 by DOMARS