MicrosoftDocs/windows-driver-docs #1301: Incomplete list of event triggers for application of new MNO settings

TheS1ngularity, 2019-02-04:
Hi,
I believe the list "What events trigger the application of new MO settings?" is incomplete because when roaming ICCID doesn't change but e.g. IMSI or Home Network does and the MNO settings also change based on COSA DB configuration.
Can you please list to what exact (and all) changes in the eSIM/SIM the Windows provisioning engine reacts to apply new MNO settings?

Thank you

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 73249166-42e2-1205-a842-818eb35eccad
* Version Independent ID: e50887fc-f2b1-9e46-7747-c32f52de952f
* Content: [COSA overview - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/cosa-overview)
* Content Source: [windows-driver-docs-pr/mobilebroadband/cosa-overview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/mobilebroadband/cosa-overview.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-02-04:
Hi @TheS1ngularity, thanks for the question. I don't know the exact answer to your question as far as an exhaustive list of event triggers goes, but let me email some folks to see if there's anything we can add to the docs here.

duncanmacmichael, 2019-03-13:
Hi @TheS1ngularity, I've gotten some clarification on this for you. According to the project manager for this feature, COSA only kicks in when the physical SIM or eSIM equivalent changes, or when the device reboots. So, the doc is in fact complete since the ICCID is identifier for the physical SIM/eSIM profile. Hope this helps.

closed 2019-03-13 by duncanmacmichael