MicrosoftDocs/windows-driver-docs #1305: WDK install

saditham, 2019-02-05:
I installed VS2017 Community Edition version 15.9.6 and followed that with WDK isntallation.  At the end of the WDK install it mentions WDK extension needs to be installed in VS2017.  How does one get it?  I would have thought the WDK install should have taken care of it.  Or documentation should have a clear link (or as part of install, better) to the extension

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2019-02-05:
Hi @saditham.

Thanks for reaching out to the Windows Driver Developer Documentation team. 

In Step 2, the page says: New as of 1709 release: The WDK installation will by default install the WDK Visual Studio extension. This must be done in order for WDK VS integration to work.

I have not heard of this failing during installation. You might want to try uninstalling and reinstalling.
Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.

You may have success posting your question in the following forums, where other driver developers and Microsoft  engineers can be found. The communities there are quite responsive.
[Windows Hardware WDK and driver development forum](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk)
[OSR NTDEV forum](http://www.osronline.com/cf.cfm?PageURL=showlists.cfm?list=NTDEV)

Again, thanks for reaching out. Best of luck with this issue.

Eliot

closed 2019-02-05 by EliotSeattle

saditham, 2019-02-06:
thanks.  But the point is -
[image]

It appears WDK install is stand-alone and integration with VS needs this extension.  Documentation is not clear and also where one may find these extensions.  Even from VS I didn't find a way to install it (at least first search did not yield it).  Recommend putting a link to it instead of just plain verbage.