MicrosoftDocs/windows-driver-docs #1307: Dead HID link

cjbrowne, 2019-02-06:
The link to the HID section points to an old msdn article and therefore redirects to a generic landing page now.

EliotSeattle, 2019-02-07:
Hi @cjbrowne.
Since you created this issue as a general GitHub issue, it's not tied to a page. Can you tell me what page you found the broken link on? I'm happy to fix that. Or you can edit the page and send a pull request. Either way.
Thanks,
Eliot

cjbrowne, 2019-02-07:
https://docs.microsoft.com/en-gb/windows-hardware/drivers/device-and-driver-technologies

I created the issue using the form on that page, I assumed it would automatically link to the page. Sorry!

EliotSeattle, 2019-02-07:
Fixed it so the link points to https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/. Should be live shortly. Thanks for the heads up, @cjbrowne

closed 2019-02-07 by EliotSeattle