MicrosoftDocs/windows-driver-docs #1311: Pattern requires quotes for [d|q]word

bendono, 2019-02-08:
Under the Pattern parameter, it states: "You can specify different types of memory in Type. If you specify a WORD, DWORD, or QWORD value, enclose it in single quotation marks (for example, 'Tag7')." This does not seem right. For example, suppose you are searching for the value 1 in a word. This works: s -[1]w @rsp-100 @rbp 1. However, as the type is a word, per the documentation the pattern should be wrapped in single quotes, hence: s -[1]w @rsp-100 @rbp '1'. But this does not actually work.

This should be clarified.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 770e8667-e5e5-e37f-e2f1-f39025028075
* Version Independent ID: 3227d5dc-3547-fba5-5b88-56f04f8dc8ce
* Content: [s (Search Memory) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/s--search-memory-)
* Content Source: [windows-driver-docs-pr/debugger/s--search-memory-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/s--search-memory-.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-02-21:
@bendono - The search command has some subtle/complex behaviors with compound parameters and it is always good to validate that the command is working as intended as you are doing. As people have gotten used to those behaviors, it is normally a matter of finding a pattern that works for you. The docs do not cover all of the permutations that are possible with the search command. 

By the way, a great resource for help on more complex searches is the OSR dev list.
[OSR's windbg List](https://community.osr.com/categories/windbg)

I may not be totally following what you are attempting to do so perhaps a few examples will help clarify.

I was able to find something with both quoted and unquoted parameter.

```
0:000> s -[1]w @rsp-100 L1000000 '1'  
0x00000056`9047f100
0x00000056`9047f104
0x00000056`904fe668
0x00000056`904fe9f8
0x00000056`904fed68
0x00000056`904ff3b0
0x00000056`904ff3b4
0x00000056`9057e4a8
0x00000056`9057eb1c
0x00000056`9057eba8
0x00000056`9057ef36
0x00000056`9057ef3a
0:000> s -[1]w @rsp-100 L100000 1  
0x00000056`901bf388
0x00000056`901bf3ec
0x00000056`901bf6ec
0x00000056`901bf730
```
As it is finding 1 at different locations, this would lead me to believe that they are searching for different things. Removing the 1 option to see the output we get this.

```
0:000> s -[]w @rsp-100 L1000000 '1'  
00000056`9047f100  0031 002d 0031 002d f158 9047 0056 0000  1.-.1.-.X.G.V...
00000056`9047f104  0031 002d f158 9047 0056 0000 0008 0000  1.-.X.G.V.......
00000056`904fe668  0031 0000 0000 0000 0000 0000 003a 0000  1...........:...

0:000> s -[]w @rsp-100 L1000000 1  
00000056`901bf388  0001 0000 0000 0000 0000 37be 7fff 0000  ...........7....
00000056`901bf3ec  0001 0000 00e0 3ba6 7fff 0000 0000 0000  .......;........
00000056`901bf6ec  0001 0000 5ed4 541f 7022 0000 ffff fffe  .....^.T"p......
```
The quoted searches for the ASCII 31 and the non quoted searches for hex 0001

I can also specify the -a option to search for ascii 31 like this: 

```
0:000> s -a @rsp-100 L1000000 1  
00000056`9047ea99  31 12 de 54 02 00 00 67-00 04 63 00 00 00 00 12  1..T...g..c.....
00000056`9047f100  31 00 2d 00 31 00 2d 00-58 f1 47 90 56 00 00 00  1.-.1.-.X.G.V...
```
As 1 is a flag and w has two meanings the parsing of compound commands that you are working with becomes complex.  And "w" may be being interpreted not as a word type, but rather as writable memory, as in this example in the docs:

```
s -[wl4]sa 0000000140000000 L400 
```
This example also shows the trailing sa option is another tool to work with.

On the docs wording around types, I believe these examples shows the use of a single quoted search string with a type parameter.

```
0:000> s -d @rsp-100 L1000000 '1'  
00000056`904fe668  00000031 00000000 00000000 0000003a  1...........:...

0:000> s -w @rsp-100 L1000000 '1'  
00000056`9047f100  0031 002d 0031 002d f158 9047 0056 0000  1.-.1.-.X.G.V...
00000056`9047f104  0031 002d f158 9047 0056 0000 0008 0000  1.-.X.G.V.......
```

So I believe that the docs are generally true, but in some more complicated uses of the search commands there can be exceptions. Better wording would probably be to say, that you may need to add quotes, depending on the other options specified, as it can be a bit more complex as you have shown with your example.

DOMARS, 2019-02-21:
@bendono - I updated the wording to clarify when quotes could be used. The changes should be live in a day or so.

DOMARS, 2019-06-07:
@bendono - Changes/updates are live so closing this issue.

closed 2019-06-07 by DOMARS