MicrosoftDocs/windows-driver-docs #1315: Is system restore point important for provisioning

csvital, 2019-02-10:
Hi, 

after two restarts on target machine at the final step of automatic provisioning on visual studio (host machine) I get this error

Creating system restore point
    ERROR: Task "Creating system restore point" failed to complete successfully.

Is it important or can I ignore it?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ce050ea3-d61e-830c-3038-f725eb4c29e1
* Version Independent ID: eb7fc65e-bcfe-dc86-3f8c-336d20eb0c86
* Content: [Provision a computer for driver deployment and testing (WDK 10) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/provision-a-target-computer-wdk-8-1)
* Content Source: [windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

DOMARS, 2019-02-11:
@csvital - Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.
You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.
•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

A few ideas on what you are running into with the restore points. You could try to manually create a restore point using control panel. If it doesn't work, look in the system logs to determine why.
The test PC should not be one that has anything important on it, as you may need to reinstall the OS during testing.
In my experience, as you will need to test installing the driver at some point anyway, I would not recommend this method of deployment using VS. I would create an INF file and use device manager to install on the target machine. I would use the KDNET utility to configure the debugger.
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/setting-up-a-network-debugging-connection-automatically
Good luck in your development efforts.

closed 2019-02-11 by DOMARS

vikaspachisia, 2020-05-14:
Even I got the same error. It is the common issue that we see when we follow the documentation.
However, I was doing it on a VM and as mentioned in another comment VMs are not supported.

But the problem I got into is the automatic configuration did the steps up to the restore point and created a "WDK Remote User" account on the test VM machine and then failed. Now, I don`t have the password for WDK Remote User Account. 

From where can I get that? 
And how can I manually configure the target system?

EliotSeattle, 2020-05-15:
Hi @vikaspachisia 

I found a [StackOverflow](https://stackoverflow.com/questions/27640883/wdkremoteuser-password) post that describes the solution neatly:

Log in to the target computer with an admin account. Go to 'Edit Local Users and Groups'.  You should see the 'WDKRemoteUser' in the list. Right-click the user and 'Set Password...'. Put in a password that will work  with your group password policy. Sign out and login as ".\WDKRemoteUser" with your current password. Retry the provisioning with the corresponding debug settings. It will run through the provisioning correctly this time.

Hope this helps.
Eliot

vikaspachisia, 2020-05-15:
Thank you so much for the response.
I am aware of this process but was looking for an official answer from Microsoft.