MicrosoftDocs/windows-driver-docs #1317: Cannot find file

rob18767, 2019-02-11:
.load "C:\Program Files (x86)\Windows Kits\10\Debuggers\x64\winext\wdfkd.dll"

Scuppered at the first hurdle again. I can see the dll at the above location however WindBg says it cannot find it. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e0f8212b-dc5d-fe98-f41e-4340ee1c74e0
* Version Independent ID: 33e7ed26-9aff-9890-f10a-fd284b7c233c
* Content: [Using the Framework's Event Logger - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/using-the-framework-s-event-logger)
* Content Source: [windows-driver-docs-pr/wdf/using-the-framework-s-event-logger.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdf/using-the-framework-s-event-logger.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-04-30:
@DOMARS any idea why a `.load` on a valid path would fail?

DOMARS, 2019-05-01:
As you might imagine, extensions may not load for a number of reasons. A few ideas:
- wdfkd should be in the default path so I would try the load command without the path like this:

` .load   wdfkd.dll`
Use the .chain command to see which extensions are already loaded. 
`.chain`
When it loads OK, no message is displayed.
Use the !wdfkd.help command to validate that it is loaded.

If you are starting out with the debugger, you may want to post questions to one of the forums, such as 

- May need to match 32/64 bit versions
- May need to be running as an administrator or check permissions are OK for this directory.

DOMARS, 2019-05-01:
If you are starting out with the debugger you may want to post to one of the forums.
[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

DOMARS, 2019-05-01:
I'm running WinDbg  Preview so my paths are in the Apps dir, but the output of .chain shows the extension being loaded. WinDbg Preview also offers a logging window that can be useful for debugging.
```
||2:0: kd> .chain
Extension DLL search Path:
...

Files\WindowsApps\Microsoft.WinDbg_1.1903.26001.0_neutral__8wekyb3d8bbwe\amd64\winext\wdfkd.dll]
    dbghelp: image 10.0.18865.1001, API 10.0.6, 
        [path: C:\Program Files\WindowsApps\Microsoft.WinDbg_1.1903.26001.0_neutral__8wekyb3d8bbwe\amd64\dbghelp.dll]
    ext: image 10.0.18865.1001, API 1.0.0, 
        [path: C:\Program Files\WindowsApps\Microsoft.WinDbg_1.1903.26001.0_neutral__8wekyb3d8bbwe\amd64\winext\ext.dll]
    exts: image 10.0.18865.1001, API 1.0.0, 
        [path: C:\Program Files\WindowsApps\Microsoft.WinDbg_1.1903.26001.0_neutral__8wekyb3d8bbwe\amd64\WINXP\exts.dll]
    kext: image 10.0.18865.1001, API 1.0.0, 
        [path: C:\Program Files\WindowsApps\Microsoft.WinDbg_1.1903.26001.0_neutral__8wekyb3d8bbwe\amd64\winext\kext.dll]
    kdexts: image 10.0.18865.1001, API 1.0.0, 
        [path: C:\Program Files\WindowsApps\Microsoft.WinDbg_1.1903.26001.0_neutral__8wekyb3d8bbwe\amd64\WINXP\kdexts.dll]
```

tedhudek, 2019-05-01:
Thanks @domars

closed 2019-05-01 by tedhudek