MicrosoftDocs/windows-driver-docs #1318: PropPageExtensions is no longer exist in sysvad

csvital, 2019-02-11:
After finished deployment of sysvad I'm looking for the System Effects Sample tab for enabling delay or swap apo as provided in documentation, but it was removed 

on 7 Jun 2017 
in commit e98f4cac93ec65a484be77d8a795538b02dab72e 
1. Removed property page extension to make Sysvad universal driver. 

from the windows-driver-samples repo.

So, is it possible to have UI for universal drivers? I'm trying to test/debug delay or swap apo in tabletaudiosample

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 59337eae-4e3a-b9c0-a90d-c2305bb5e7ca
* Version Independent ID: 6bd7143b-e8b8-981a-3b9b-1814ef761a43
* Content: [Implementing a UI for Configuring APO Effects - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/implementing-a-ui-for-configuring-apo-effects)
* Content Source: [windows-driver-docs-pr/audio/implementing-a-ui-for-configuring-apo-effects.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/implementing-a-ui-for-configuring-apo-effects.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-02-21:
@csvital - Thanks for pointing this out. I downloaded the sample and the UI project does indeed seem to be missing. I'm checking with the dev team to see what I can find out about this.

DOMARS, 2019-10-08:
@csvital - I worked with the dev team and have added this note to the top of the topic.

*This customization is no longer supported after Windows 10, version 1809 and the PropPageExtensions project is no longer present in the Sysvad Sample. For later versions of Windows, the recommended approach is to create a Hardware Support App. For more information, see Hardware Support App (HSA): Steps for Driver Developers.*

Thanks for pointing out this issue and for your patience in tracking down a recommended UI replacement for APO configuration.

closed 2019-10-08 by DOMARS