MicrosoftDocs/windows-driver-docs #1319: need update?

minhyuk, 2019-02-11:
Hello, I am confused about process attestation with this document. 
because it does not sync the current GUI of the partner center.
and you mentioned attestation process don't need CAT(catalog file), but it failed when submitting without CAT.
would you please update?

---
#### 문서 세부 정보

⚠ *이 섹션을 편집하지 마세요. docs.microsoft.com ➟ GitHub 문제 연결에 필요합니다.*

* ID: ed287852-04b7-1569-fd51-2640cfa24f57
* Version Independent ID: a8ce732f-5b41-f1ce-e37d-a7c2794546a4
* Content: [Attestation signing a kernel driver for public release - Windows drivers](https://docs.microsoft.com/ko-kr/windows-hardware/drivers/dashboard/attestation-signing-a-kernel-driver-for-public-release)
* Content Source: [windows-driver-docs-pr/dashboard/attestation-signing-a-kernel-driver-for-public-release.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/attestation-signing-a-kernel-driver-for-public-release.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

GabeSJones, 2019-03-08:
I disagree that CAT files are required.  We never put CAT files in our CABs for attested signing.  We just dump in the .SYS and .INF files and everything works fine.

tedhudek, 2020-11-03:
Need to confirm that screenshot in https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/attestation-signing-a-kernel-driver-for-public-release#submit-the-ev-signed-cab-file-using-the-partner-center is out-of-date

tedhudek, 2020-11-10:
Thanks for the feedback and apologies for the delay.  I've just pushed d518c20 with an updated image.  The fix should appear on the Docs site within 10 minutes.

closed 2020-11-10 by tedhudek