MicrosoftDocs/windows-driver-docs #1321: Missing ASL compiler download

rgough71, 2019-02-13:
It would be nice if you could include a link whereby the MS ASL compiler can be downloaded.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 146c2e57-a40b-33e0-c108-37f7555eda8a
* Version Independent ID: 81b0b392-c18a-f083-17fa-7e6dca629a52
* Content: [Microsoft ASL compiler - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/microsoft-asl-compiler#additional-resources)
* Content Source: [windows-driver-docs-pr/bringup/microsoft-asl-compiler.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/microsoft-asl-compiler.md)
* Product: **windows-hardware**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

kyanha, 2019-04-21:
The parent to this page states:

```
Version 5.0 of the Microsoft ACPI source language (ASL) compiler supports the features in the Advanced Configuration and Power Interface Specification, Revision 5.0 (ACPI 5.0 specification). The ASL compiler is distributed with the Windows Driver Kit (WDK) 8.1.
```

I agree, it would be nice to have a secondary statement on this page that it can be downloaded as part of that kit.  Is it also part of WDK 10?

suirsuss, 2019-04-28:
No joke man. Without a link and trying to troubleshoot my dear and brand new eGPU, I have little patience for reading! Thanks to kyanha for answering. Now I know.

barrygolden, 2020-12-18:
Thanks for reaching out to the Windows Driver Developer Documentation team with your comments about a download link for the ASL complier.

This topic has been updated with information on downloading the Windows Driver Kit (WDK) that includes the ASL compiler.

johnothwolo, 2023-04-19:
I have WDK 10.22000.832 fully installed on Windows 11. The path "C:\Program Files (x86)\Windows Kits\10\Tools\" does not exist. Please update the documentation if the ACPI tools were discontinued. 

You have no idea how frustrating it is when all the internet sources say to download WDK and use asl.exe, only to find that the directory doesn't even exist... on multiple WDK versions.

tedhudek, 2023-04-19:
Hi @johnothwolo sorry for the trouble.  I just searched my recent kit installation and I see `asl.exe` in `C:\Program Files (x86)\Windows Kits\10\Tools\10.0.25330.0\arm\ACPIVerify`.  Is that what you're looking for?

johnothwolo, 2023-04-19:
Yeah, so this is my Windows Kits directory:

[image]

And these are the kits I have installed:

[image]

Idk why, but I don't have a `Tools` directory.

barrygolden, 2023-04-19:
@johnothwolo, I think you have the SDK installed correctly as shown above.

You have not yet installed the WDK (Windows Driver Kit) which will install the asl.exe tool you need.

The SDK and the WDK are two separate kits. You do need to (1) install the SDK as a precursor to the WDK. Then you need to (2) install the WDK.

The WDK can be downloaded here: 

https://learn.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk#download-icon-step-3-install-windows-11-version-22h2-wdk

barrygolden, 2023-04-19:
@johnothwolo, when the WDK is installed, you should see something similar to the following for your SDK and WDK install points:

[image]

I'm showing Insider Preview versions above, but you should have a similar SDK and WDK version that match (ignore the fonts entry as not relevant here).

tedhudek, 2023-04-29:
Updated the location of asl.exe on the doc page.  This update should reach the live site momentarily.

Thanks again for the feedback!

closed 2023-04-29 by tedhudek

johnothwolo, 2023-05-01:
> @johnothwolo, when the WDK is installed, you should see something similar to the following for your SDK and WDK install points:
> 
> [image]
> 
> I'm showing Insider Preview versions above, but you should have a similar SDK and WDK version that match (ignore the fonts entry as not relevant here).

Sorry for not updating you, it was kind of embarrassing when I realized my mistake. I downloaded and installed the right package; I confused the WDK with the SDK.

tedhudek, 2023-05-01:
No worries John, thanks for letting us know.  Glad you got it working.