MicrosoftDocs/windows-driver-docs #1332: Installing a Universal Windows Driver link is broken

sarman1998, 2019-02-15:
The link in the See Also section takes me to https://docs.microsoft.com/en-us/windows-hardware/drivers/
I looked for the correct link but i couldn't find it here.  I did see it in someones old fork of the docs on GitHub so i suspect that section was renamed during some rewrite.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: db23c35e-03c3-2d04-bb25-dc57082606d1
* Version Independent ID: 539d8683-17aa-2644-020f-df4c6373d872
* Content: [Using a Universal INF File - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-a-universal-inf-file)
* Content Source: [windows-driver-docs-pr/install/using-a-universal-inf-file.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/using-a-universal-inf-file.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-04-30:
@billlattimer please update the first link in the See Also

billlattimer, 2019-11-01:
@sarman1998 thank you for catching this; we've redirected the link to the "Getting Started with Universal Windows Drivers" as the best match.

closed 2019-11-01 by billlattimer