MicrosoftDocs/windows-driver-docs #1334: Possible missing parameter 1 value

hunger1, 2019-02-16:
I see the following event in the Event Viewer of my Windows 10 Pro 1803 system:

The computer has rebooted from a bugcheck.  The bugcheck was: 0x000000e6 (0x0000000000000026, 0xffffd08bb1435060, 0x0000000000000040, 0x0000000000000006).

The first parameter, 0x26, is not shown in the list above.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b26a9559-aee1-4875-0060-d713a80133dd
* Version Independent ID: 27ed6e22-357d-3d01-16c0-eaa727e5ec1d
* Content: [Bug Check 0xE6 DRIVER_VERIFIER_DMA_VIOLATION - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xe6--driver-verifier-dma-violation)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xe6--driver-verifier-dma-violation.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xe6--driver-verifier-dma-violation.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

donmac2, 2019-02-16:
I have the same problem. In case it's not linked, the webpage he's referring to is at https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xe6--driver-verifier-dma-violation

WeaselDev, 2019-02-27:
Same issue here with Windows 10. 
The bugcheck was: 0x000000e6 (00000000`00000026, ffff950d`53459060, 00000000`00000027, 00000000`00000006).

garysheppardjr, 2019-02-28:
Mine is almost identical to the others posted:

`0x000000e6 (0x0000000000000026, 0xffffd48b9ae8d060, 0x0000000000000024, 0x0000000000000006)`

ralish, 2019-03-05:
For what it's worth WinDbg maps `0x26` to the following description: *IOMMU detected DMA violation*.

Here's the header of a bugcheck analysis with the same bugcheck code and first argument:
```
DRIVER_VERIFIER_DMA_VIOLATION (e6)
An illegal DMA operation was attempted by a driver being verified.
Arguments:
Arg1: 0000000000000026, IOMMU detected DMA violation.
Arg2: ffffb90c3e547060, Device Object of faulting device.
Arg3: 0000000000000025, Faulting information (usually faulting physical address).
Arg4: 0000000000000006, Fault type (hardware specific).
```

stozk, 2019-03-08:
Same problem here with parameter 1 26 since 2 or 3 days. Seems like some driver or update broke something.

DOMARS, 2019-03-14:
@hunger1 - I will check with the dev team to work on updates here. 
In the meantime don't forget the use of !analyze to display bug check information, like this:

```
0: kd> !analyze -show 0xe6 0x26
DRIVER_VERIFIER_DMA_VIOLATION (e6)
An illegal DMA operation was attempted by a driver being verified.
Arguments:
Arg1: 0000000000000026, IOMMU detected DMA violation.
Arg2: 0000000000000000, Device Object of faulting device.
Arg3: 0000000000000000, Faulting information (usually faulting physical address).
Arg4: 0000000000000000, Fault type (hardware specific).
```

stozk, 2019-03-14:
In my case the ASMedia driver caused this. The latest driver introduced some new setting:

HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Services\asmtxhci\Parameters
"DmaRemappingCompatible"

Uninstalling the driver doesn't help because the key will stay in the registry. 
Deleting the key fixed the bluescreens.

BR

DOMARS, 2019-04-15:
@hunger1 - The topic now includes an updated list of parameters. Thanks for your input on this stop code.

closed 2019-04-15 by DOMARS