MicrosoftDocs/windows-driver-docs #1342: WDK forces /Qspectre for new projects.  Uninstall of extension prevents VS2017 uninstall or repair!

gilgm12, 2019-02-23:
[image]
[image]

My VS2017 workspaces are hosed now. :(

EliotSeattle, 2019-02-25:
Hi @gilgm12 
Sorry that you're having this happen. I have not seen this occur before. Do you have any more repo details?
Also, my team handles feedback about the documentation. Is there an error in the docs? If this is really about the product, you'll have better success submitting feedback through Feedback Hub (Click **Start**, and then type **Feedback Hub**).

EliotSeattle, 2019-03-06:
Closing without a response to question and b/c this is product feedback.

closed 2019-03-06 by EliotSeattle