MicrosoftDocs/windows-driver-docs #1343: Possible values for parameter 1 and other parameters meaning

LuigiBruno, 2019-02-25:
Are there other possible values for parameter 1? I've sen an example of this bug check submitted by one user in TechNet forums having the value "0xffffffffc0000005" (which resembles the NT code for memory access violation).
Also, what is the meaning of the other 3 parameters?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5d544beb-8680-03d5-37ce-76881997ffe6
* Version Independent ID: fa5e1d25-0e5c-8207-0e84-6cfd93ee60f0
* Content: [Bug Check 0x126 NETIO_INVALID_POOL_CALLER - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x126--netio-invalid-pool-caller)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x126--netio-invalid-pool-caller.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x126--netio-invalid-pool-caller.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-27:
@LuigiBruno  - Thanks for the input, that we should add additional information on Bug Check 0x126. 

We use customer input like this as well as data on how often the bug checks occur, to determine how we allocate our documentation team resources.

If you happened to have output from !analyze for the four parameters I would be interested in seeing that, as this seems to be a pretty rare bug check.

DOMARS, 2020-12-18:
@LuigiBruno - I looked at this a bit, and it doesn't seem that that a value of  "0xffffffffc0000005" would make any sense. I believe param 1 should be a single digit. So perhaps they confused the param value of 1 with one of the other params, such as 2 or 3.  Those could be a memory range pointer and the value of "0xffffffffc0000005" makes more sense there. 
As I have mentioned, we use input like this to determine what topics we work on. So thanks again for the input.

closed 2020-12-18 by DOMARS