MicrosoftDocs/windows-driver-docs #1352: unclear whether EV certs are required for partner center or not

psifertex, 2019-02-26:
I'm having trouble using a standard certificate (that validates with `signtool verify /pa`) for the Windows 10 desktop application partner page: https://partner.microsoft.com/en-US/dashboard/Account/CertificateUpload

The documentation on this page is ambiguous on the topic. First, it says: 

`The Partner Center will enforce mandatory EV certificates for submissions later this year.`

But right below it says: 

```Standard Code Signing
Provides standard level of identity validation
Requires shorter processing times and lower cost
Can be used for all Partner Center services except LSA, and UEFI file signing services.
```
I'm trying to verify a standard Windows 10 application, not a driver so it seems like a standard cert should be sufficient. 

On a related note, when I try to use the signing page, it errors out and says:

```Error: The code signing certificate authority (CA) you used isn’t on the approved list of root certificates. Please get a certificate from one of the CAs listed on the certificate info page.``

Unfortunately the link it includes for "cerificate info" is: https://partner.microsoft.com/en-US/dashboard/Account/CertificateInfo which just 302 redirects to the same page which doesn't specify any information at all about CA requirements. I'd have assumed the standard CA list listed on this page would be sufficient but maybe that's not the case? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 93d3f097-d81b-66c2-4835-d63758687d34
* Version Independent ID: fe9d1cdd-966e-89b5-c66b-60531b0ec655
* Content: [Get a code signing certificate - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-a-code-signing-certificate)
* Content Source: [windows-driver-docs-pr/dashboard/get-a-code-signing-certificate.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-a-code-signing-certificate.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-02-26:
Thanks, @psifertex, for bringing this up. That doc needed updating. We've made changes, and it will go live later today. In short, desktop Win32 application signing is no longer done. See [Win32 logo certification deprecation](https://blogs.msdn.microsoft.com/windows_hardware_certification/2018/04/19/win32-logo-certification-deprecation/) for the summary, until the updated page loads.

closed 2019-02-26 by EliotSeattle

psifertex, 2019-02-26:
I'm not sure that answers my question -- the link is talking about Pre Windows 10 32bit applications if I read it correctly. I'm interested in telemetry and signing for a 64bit Win10 desktop application. Or am I misunderstanding?

psifertex, 2019-02-26:
Specifically, I'm looking for crash dumps from users of my software that is a 64bit PE signed with a Comodo reseller cert. 

I believe this used to be available on sysdev and was under the impression the new link above was the place to go for it but I've never used sysdev myself, just from what I can read online.

EliotSeattle, 2019-02-26:
Oh, sorry, @psifertex. I misunderstood your request.

Our docs are specifically about the hardware and driver certification process. I think you will find what you need on the [Windows Desktop Application Program](https://docs.microsoft.com/en-us/windows/desktop/appxpkg/windows-desktop-application-program)

First, you'll need a Partner Center Account (I'm not positive if you need an EV cert for this). You may have to contact Support for your issue, using the steps at [https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq). Note that you need to be signed in with the same account you ask questions about (that’s why you should create the Partner Center Account).

Hope this helps.
Eliot

psifertex, 2019-02-27:
Thanks @EliotSeattle, sorry for confusing the two pages.

Yes, that's the process I'm following now that's broken. I don't see a link on that page however to point out a problem with the documentation -- specifically the broken link on the error message along with a lack of any actual information about which CAs are required. I have a Partner Center Account (the account itself required no EV cert). I'll try the email address that eventually comes up following the support link you posted, thanks.

Framnk, 2019-12-30:
@psifertex I'm curious if you ever received an answer to this as we are experiencing issues with Partner Center analytics and I'm wondering if they silently added the requirement for EV certificates.

psifertex, 2019-12-31:
No, I never resolved it, I simply stopped trying to run it down and gave up on getting the dumps.