MicrosoftDocs/windows-driver-docs #1355: Missing parameter 1 value = 0x5010

winter3angle, 2019-02-27:
Corresponding bugcheck was hit during WHQL "Trim Test" on Windows 7 with enabled verifier on our own drivers, ataport.sys, msahci.sys and msahci_hvt.sys

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-01:
@winter3angle - This doesn't seem to be feedback on the docs, so closing this issue. For community comments see the forums, for example: 
•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

closed 2019-03-01 by DOMARS

winter3angle, 2019-03-01:
@DOMARS could you please kindly clarify why this isn't related to the docs? I saw a couple of similar issues that were addressed (eg. #660). The issue is not about the bugcheck itself, but about the undocumented parameter 1 value.