MicrosoftDocs/windows-driver-docs #1356: Exception code "0x80000003"

LuigiBruno, 2019-02-27:
The list of NTSTATUS values states that "0x80000003" stands for "{EXCEPTION} Breakpoint A breakpoint has been reached.", but I've analyzed a memory dump file where WinDbg reported the below information

EXCEPTION_CODE: (HRESULT) 0x80000003 (2147483651) - One or more arguments are invalid

Which is the correct value for that status code?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b19eb27e-40b0-c765-d826-8b9962e4c1e6
* Version Independent ID: f9231818-2d1d-be08-7ce2-92b97edd5613
* Content: [Bug Check 0x3B SYSTEM_SERVICE_EXCEPTION - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x3b--system-service-exception)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x3b--system-service-exception.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x3b--system-service-exception.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-01:
@LuigiBruno - I believe NTSTATUS values are different from HRESULT values. See the protocol docs:
https://docs.microsoft.com/en-us/openspecs/windows_protocols/ms-erref/705fb797-2175-4a90-b5a3-3918024b10b8

LuigiBruno, 2019-03-01:
Thanks @DOMARS.

DOMARS, 2019-03-11:
@LuigiBruno - Your welcome.

closed 2019-03-11 by DOMARS