MicrosoftDocs/windows-driver-docs #1361: Broken links for SDK/WDK preview

HoShiMin, 2019-02-28:
SDK and WDK link is the same and broken: https://www.microsoft.com/software-download/windowsinsiderpreviewWDK

---
#### Сведения о документе

⚠ *Не вносите правки в этот раздел. Это необходимо для связывания страницы сайта docs.microsoft.com с вопросом на GitHub.*

* ID: 34d06a09-bca8-3177-00fd-2195f56c4fb9
* Version Independent ID: 5328b553-ad08-4808-0b14-9720484040ce
* Content: [Installing preview versions of the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/ru-ru/windows-hardware/drivers/installing-preview-versions-wdk)
* Content Source: [windows-driver-docs-pr/installing-preview-versions-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/installing-preview-versions-wdk.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-02-28:
Hi @HoShiMin.
Thanks for reaching out. I tried every link on [Installing preview versions of the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/windows-hardware/drivers/installing-preview-versions-wdk), and they all seem to work for me. Yes, the link is the same for the WDK and the SDK, for both downloads are found on the same destination page. But I find no broken links. Can you tell me specifically which links you're talking about?

Cheers,
Eliot

HoShiMin, 2019-02-28:
Sure, but I've check it in Russian localization, and I already found an en-us version that works fine.
https://www.microsoft.com/ru-ru/software-download/windowsinsiderpreviewWDK

EliotSeattle, 2019-03-06:
Looked into this more, @HoShiMin,
[image]
 but I can't reproduce any error. When I set my browser to ru-ru locale, the links to the windowsinsiderpreviewWDK download work fine, though they don't specify a locale for the download. We do this on purpose, to guard against trying to go to a page that's not localized. If you manually add the ru-ru locale to the URL, you will get a 404, because we don't have that page in Russion. I've attached a screenshot to show my experience. Please let me know if I'm missing something, but I'm going to go ahead and close this.
Thanks for reaching out.
Eliot

closed 2019-03-06 by EliotSeattle