MicrosoftDocs/windows-driver-docs #1366: Document symchk /ov

KalleOlaviNiemitalo, 2019-03-01:
`symchk /?` describes:

```
/ov       Print version information for checked binaries as well.
```

Please document the `/ov` option here.

In symchk.exe version 10.0.10240.16384, `/ov` used to display the FILEVERSION of each file. In symchk.exe version 10.0.17763.132, it now displays the PRODUCTVERSION instead. This seems worth documenting, too.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1f802ab4-103b-40bd-7aff-9b519b2e9f56
* Version Independent ID: 0d3ae05a-41f8-7cea-c9a8-25937b235905
* Content: [SymChk Command-Line Options - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/symchk-command-line-options#feedback)
* Content Source: [windows-driver-docs-pr/debugger/symchk-command-line-options.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/symchk-command-line-options.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-13:
@KalleOlaviNiemitalo - Thanks for pointing out this omission. I added /ov and also /ol. Updates should be live in a day or two.

KalleOlaviNiemitalo, 2019-03-14:
How about documenting the change from FILEVERSION to PRODUCTVERSION?

DOMARS, 2019-03-18:
@KalleOlaviNiemitalo - Fixes are live. For now I think I will parallel the general description of the version information that is displayed in the /?, but thanks for the suggestion.

closed 2019-03-18 by DOMARS