MicrosoftDocs/windows-driver-docs #1369: The link to "test categories" does not work

jurva, 2019-03-02:
In addition, the link to Hardware Certification Kit (HCK) points to an outdated page.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: db0c1e7f-3231-e9d5-be1d-0053815ee006
* Version Independent ID: 730bbc6f-afa2-1876-a466-14c0931b56a1
* Content: [PnP Device Installation Signing Requirements - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/pnp-device-installation-signing-requirements--windows-vista-and-later-)
* Content Source: [windows-driver-docs-pr/install/pnp-device-installation-signing-requirements--windows-vista-and-later-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/pnp-device-installation-signing-requirements--windows-vista-and-later-.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-04-25:
The link should point to [Windows Hardware Lab Kit](https://docs.microsoft.com/windows-hardware/test/hlk/).  @billlattimer can you please update?

billlattimer, 2019-04-25:
Updated and should go live shortly - thanks, @jurva !

closed 2019-04-25 by billlattimer