MicrosoftDocs/windows-driver-docs #1370: Incorrect character escaping in code samples

ChrisMaddock, 2019-03-03:
It looks like a number of HTML characters have been incorrectly escaped in code samples in this repository.

For example: apostrophe's showing as `&#39;` in the code comments here:
https://docs.microsoft.com/en-us/windows-hardware/drivers/install/checking-for-in-progress-installations

And `&quot;` in the code samples here:
https://github.com/MicrosoftDocs/windows-driver-docs/blob/b9be32d9e5217219dee224527d32979d396a9027/windows-driver-docs-pr/develop/debugging-a-universal-driver.md

EliotSeattle, 2019-03-04:
Thanks, @ChrisMaddock.
I avow, once more, that I never met a migration tool I didn't like.
Got these. They'll be live shortly.  In the future, please feel free to create a pull request if you see anything like this.
Cheers,
Eliot

ChrisMaddock, 2019-03-04:
Hey @EliotSeattle - sorry, my bug report wasn't clear.

The above were just two examples - this same problem happens in 100s of places across this repo. That's one of the reasons I didn't pull request, was more than I was able to commit to.

Would you be able to reopen this issue, for the remaining problems? Here's a search showing problems with the apostrophe escaping - the same can be found for other HTML escaping characters too.

https://github.com/MicrosoftDocs/windows-driver-docs/search?q=%22%26%2339%3B%22&unscoped_q=%22%26%2339%3B%22

EliotSeattle, 2019-03-04:
Oh, you mean like real work? Got it. :-(
Do you happen to have a list of the other escape characters you've seen? It'd be helpful. If not, we can come up with some regular expression searches.

ChrisMaddock, 2019-03-04:
Ikr. 😞 

I can see pretty broken cases of much everything common... `&amp; &gt; &lt; &quot;` etc. I haven't searched much for other numeric escapes.

tedhudek, 2021-01-21:
Not repro'ing this problem now.  I do see instances of `&gt;`, for instance, but they render fine, as the greater than symbol.

Appreciate the report.

closed 2021-01-21 by tedhudek