MicrosoftDocs/windows-driver-docs #1371: Link broken

jurva, 2019-03-03:
Microsoft Cross-Certificates for Windows Vista Kernel Mode Code Signing link is broken. Points here:
https://www.microsoft.com/whdc/winlogo/drvsign/crosscert.mspx
"We are sorry, the page you requested cannot be found."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: fe1729cb-4c11-c658-7c0b-c7d19e3e0340
* Version Independent ID: 61f4cda3-d90d-1619-fa0f-4fd3dabfc44a
* Content: [Obtaining a Software Publisher Certificate (SPC) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/obtaining-a-software-publisher-certificate--spc-)
* Content Source: [windows-driver-docs-pr/install/obtaining-a-software-publisher-certificate--spc-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/obtaining-a-software-publisher-certificate--spc-.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

dollerbill, 2019-04-25:
Issue remains 4/25/19

tedhudek, 2019-05-01:
Thanks @jurva I've made the fix but due to some internal issues this won't appear on the live site until Monday.  In the meantime, the link that you want is [Cross-Certificates for Kernel Mode Code Signing](https://docs.microsoft.com/windows-hardware/drivers/install/cross-certificates-for-kernel-mode-code-signing).

closed 2019-05-01 by tedhudek