MicrosoftDocs/windows-driver-docs #1373: Driver-updatable device FW,

huagj, 2019-03-04:
As storage device can update their FW in windows OS, 
How can we release storage device FW through WU and update their FW in windows OS but not need UEFI support?
Driver-updatable device firmware.

This device firmware can be updated by the device driver during the normal Windows OS runtime. Updating device firmware using normal Windows OS drivers is not covered by this paper.

---
#### 文档详细信息

⚠ *请勿编辑此部分。 docs.microsoft.com ➟ GitHub 问题链接必须具有此部分。*

* ID: 1b4320a0-085e-4ca1-e83a-1b030b480100
* Version Independent ID: 5e474f7e-998e-1d76-49e7-fbc67ca1071d
* Content: [Windows UEFI firmware update platform - Windows drivers](https://docs.microsoft.com/zh-cn/windows-hardware/drivers/bringup/windows-uefi-firmware-update-platform)
* Content Source: [windows-driver-docs-pr/bringup/windows-uefi-firmware-update-platform.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/windows-uefi-firmware-update-platform.md)
* Product: **windows-hardware**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

RAJU2529, 2019-05-29:
I  think this  article states that UEFI  firmware. But updating other device driver or firmwares for like  harddisks or ssd or camera or tpm or dvd rom .
Those device needs special type  firmwares in the form of  .bin or .dat or .exe files

barrygolden, 2020-12-18:
Thanks for reaching out to the Windows Driver Developer Documentation team. See the [Storage Firmware Update (SFU)](https://docs.microsoft.com/windows-hardware/drivers/sfu/) and [Component Firmware Update (CFU)](https://docs.microsoft.com/windows-hardware/drivers/cfu/) topics for a possible solution to your scenario.

You may have also have success posting your questions in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

[MSDN Windows Hardware WDK and Driver Development forum](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

[NTDEV forum at OSR Online](https://www.osronline.com/showlists.cfm?list=ntdev)

closed 2020-12-18 by barrygolden