MicrosoftDocs/windows-driver-docs #1374: Missing documentation for parameter 1 = 0x3453

snowkoan, 2019-03-04:
Another undocumented stop code...

MEMORY_MANAGEMENT (1a)
    # Any other values for parameter 1 must be individually examined.
Arguments:
Arg1: 0000000000003453, The subtype of the bugcheck.
Arg2: ffffac0a88f49080
Arg3: 0000000000008500
Arg4: 0000000000000003

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-14:
@snowkoan - This param does seem to occur, so I'm checking with the dev team to see if we can document this. Thanks for passing on what you are seeing with 0x1A.

DOMARS, 2019-04-15:
@snowkoan - 0x3453 is now documented. Thanks for providing input on the bug check docs.

closed 2019-04-15 by DOMARS