MicrosoftDocs/windows-driver-docs #1375: staticContentClear.xml missing closing </appcmd> element

ryan-roberts, 2019-03-05:
Without the closing "&lt;/appcmd&gt;" in "staticContentClear.xml" the following is seen during installation script execution: "ERROR ( hresult:80004005, message:Failed to obtain input for command: Unspecified error"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a0f38fcd-06ec-37b1-3503-d4e8484b6147
* Version Independent ID: a65d3ebd-31da-84b2-ca78-312f166b99f5
* Content: [SymProxy Automated Installation - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/symproxy-automated-installation#feedback)
* Content Source: [windows-driver-docs-pr/debugger/symproxy-automated-installation.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/symproxy-automated-installation.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-12:
@ryan-roberts - The sample should show well formed XML including the closing </appcmd>. 
A fix is on the way. Thanks for seeing this error.

DOMARS, 2019-03-18:
@ryan-roberts - The fix to add the include the closing "</appcmd>" is now live. Thanks again for seeing this error in our docs.

closed 2019-03-18 by DOMARS