MicrosoftDocs/windows-driver-docs #1380: SymProxy requires trusted root certificates for upstream providers like msdl.microsoft.com/download/symbols

ryan-roberts, 2019-03-05:
Please amend the SymProxy documentation with this pretty large prerequisite.

I found out the hard way that SymProxy silently fails debugger requests with 404 if the upstream provider uses HTTPS/TLS and the root certificate for the endpoint isn't in the Trusted Root store on the machine where SymProxy is running. Our servers are configured to only install a small set of root certificates and this list didn't include the "Baltimore CyberTrust Root" certificate which msdl.microsoft.com/windows.net chains up to. In this configuration SymProxy never returned symbols from the upstream Microsoft provider. I could see in Wireshark that SymProxy would query upstream but it would never use the results. No errors/warnings showed up in the SymProxy event log. After much troubleshooting it was found that simply installing the Baltimore CyberTrust Root certificate in root fixed things.

Ideally SymProxy would also log a warning/error in this scenario but that's product feedback.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 94cca664-53c7-ee65-5975-3d7a8e56bb07
* Version Independent ID: 2b130c88-3228-1ea1-c57a-20469cc6525b
* Content: [Installing SymProxy - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/installing-symproxy)
* Content Source: [windows-driver-docs-pr/debugger/installing-symproxy.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/installing-symproxy.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-12:
@ryan-roberts - This looks similar to this issue https://support.microsoft.com/en-us/help/3187516/configuration-manager-service-connection-point-doesn-t-download-update

Can you confirm my assumption that the default Server 2016 IIS would contain the "Baltimore CyberTrust Root"? In your environment, you removed those default certs, so as to reduce your exposure?

I'm happy to add a note to this topic indicating that the "Baltimore CyberTrust Root" should not be removed from the Server.

Although a somewhat different context, this topic might be relevant here for others?

https://docs.microsoft.com/en-us/iis/troubleshoot/security-issues/troubleshooting-ssl-related-issues-server-certificate

DOMARS, 2019-06-07:
@ryan-roberts - Can you confirm my assumption that the default Server 2016 IIS would contain the "Baltimore CyberTrust Root"? In your environment, you removed those default certs, so as to reduce your exposure?

ryan-roberts, 2019-06-09:
Sorry for missing your first response Don. I believe starting with Windows Server 2012 previously-available root certificates were removed from WSUS ([see this SF thread for similar discussion](https://serverfault.com/questions/541922/where-to-get-root-ca-certificates-for-windows-server-now-that-microsoft-no-longe)). So it's not a matter of the certificates being removed but no Windows Server install post WS2012 starts with any of those 3rd party certificates. In our server environment 1) policy was configured to not download root certificates (policy "_Turn off Automatic Root Certificates Update_" was Enabled) and 2) outbound network access did not allow access to the Microsoft endpoint to retrieve those certificates. To get SymProxy working we needed to disable policy described in #1 and open access to at least ctldl.windowsupdates.com.

DOMARS, 2019-06-10:
@ryan-roberts - Thanks for that clarification.  Here is the proposed wording.

Note that the certificates store that contains the access to symproxy needs to be up to date. Certificates that are normally installed with IIS and Windows server such as the "Baltimore CyberTrust Root" are used for HTTPS/TLS communication to the upstream provider, and they need to be in the Trusted Root store on the machine where SymProxy is running. For general information on troubleshooting SSL issues, see [Troubleshooting SSL related issues (Server Certificate)](https://docs.microsoft.com/iis/troubleshoot/security-issues/troubleshooting-ssl-related-issues-server-certificate).

DOMARS, 2019-06-10:
Adding @aluhrs13 for review of wording to describe required certs for symbol server https access. 

Can you confirm the use of the "Baltimore CyberTrust Root"? Is there another way we should describe the cert requirments and how we chain up?

aluhrs13, 2019-06-10:
I'm not the right person to answer that... Andrew or Scott might know more.

DOMARS, 2019-06-10:
@aluhrs13 - Thanks, I will follow up with Andrew.

DOMARS, 2019-06-20:
@ryan-roberts - Following up with a dev who looks to be out of office for a bit...

DOMARS, 2020-10-28:
@ryan-roberts - Thanks for this issue, and apologies for the delay. I have created a work item for the dev team for this work and we will be tracking this request there, so I will close this issue. I don't have an ETA for that work right now. 
If you need information on this task, these newsgroups can be helpful.

•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
•	[OSR's ntdev List](https://community.osr.com/categories/ntdev)
•	[OSR's windbg List](https://community.osr.com/categories/windbg)

Thanks again for your interest in the debugger docs.

closed 2020-10-28 by DOMARS