MicrosoftDocs/windows-driver-docs #1381: FailureTimeout and similar settings deprecated in latest Win10 SDK releases?

ryan-roberts, 2019-03-05:
It seems symproxy, starting with version 10.0.18317.1000 and through to the current release of 10.0.18342.1, no longer reads keys: FailureTimeout, FailureCount, FailurePeriod, and FailureBlackout. Is that still a supported feature in the newest releases?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8a95a56a-d534-9fe7-6b12-eccac97f2452
* Version Independent ID: 6ed02813-5670-cec5-876c-720f93ea3248
* Content: [Configuring the Registry - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/configuring-the-registry)
* Content Source: [windows-driver-docs-pr/debugger/configuring-the-registry.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/configuring-the-registry.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

ryan-roberts, 2019-03-05:
These settings appear to also be no longer read with the latest releases: LogLevel, MissFileCache, and MissFileThreads

Put another way, these appear to be the only settings read on startup: NoInternetProxy
NoLongerIndexedAuthoritive
NoFilePointers
NoUncompress
NoCache
MissTimeout
MissAgeTimeout
MissAgeCheck
RequestTimeout
RetryAppHang
UriFilter
UriTiers

ryan-roberts, 2019-03-05:
That also means the following settings aren't described here:
NoLongerIndexedAuthoritive
MissAgeTimeout
RequestTimeout
RetryAppHang
UriFilter
UriTiers

DOMARS, 2019-03-11:
@ryan-roberts - Thanks for passing on what you are seeing. How are you checking to see what entries are valid for symproxy?

DOMARS, 2019-03-12:
@ryan-roberts - I'm seeing the changes in places such as the included reg file and I'm working on updates.

ryan-roberts, 2019-04-24:
Sorry seeing your question late-- I gathered the settings differences largely from Process Monitor events captured during SymProxy startup.

DOMARS, 2019-10-04:
@ryan-roberts - Following up with the dev to see if we can get descriptions for the new entries, such as MissAgeTimeout.

DOMARS, 2019-10-24:
@ryan-roberts - Once again following up with the dev.

DOMARS, 2020-10-28:
@ryan-roberts - Thanks for this issue, and apologies for the delay. I have created a work item for the dev team for this work and we will be tracking this request there, so I will close this issue. I don't have an ETA for that work right now. 
If you need information on this task, these newsgroups can be helpful.

•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
•	[OSR's ntdev List](https://community.osr.com/categories/ntdev)
•	[OSR's windbg List](https://community.osr.com/categories/windbg)

Thanks again for your interest in the debugger docs.

closed 2020-10-28 by DOMARS