MicrosoftDocs/windows-driver-docs #1384: get-a-code-signing-certificate Comodo link points to InstantSSL

OzBob, 2019-03-06:
The 'Comodo' link has been tampered with!

The ["Microsoft link"](https://go.microsoft.com/fwlink/?linkid=863208) currently points to InstantSSL "https://www.instantssl.com/code-signing-certificates"

instead of the new owners of Comodo, Sectigo: https://sectigo.com/products/signing-certificates/code-signing

I would create a pull request myself and correct the Docs, but do not know how to create a new Microsoft shorturl.

EliotSeattle, 2019-03-06:
@SillyKeith, can you please look at this and Issue #1010

tedhudek, 2019-05-30:
Feedback applies to the Sectigo link on https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-a-code-signing-certificate

tedhudek, 2019-05-30:
Issue if fixed

closed 2019-05-30 by tedhudek