MicrosoftDocs/windows-driver-docs #1387: Missing documentation for parameter 1=0x15001

metata0201, 2019-03-08:
MEMORY_MANAGEMENT (1a)
    # Any other values for parameter 1 must be individually examined.
Arguments:
Arg1: 0000000000015001, The subtype of the bugcheck.
Arg2: ffffbd0a1da2aa80
Arg3: ffffbd0a1e98e440
Arg4: 0000000000000000

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-14:
@metata0201 - Checking with the dev team to see if we can document this.

DOMARS, 2019-04-15:
@metata0201 - 0x15001 is now documented. Thank you for providing input on the bug check documentation.

closed 2019-04-15 by DOMARS