MicrosoftDocs/windows-driver-docs #1394: License for the C# sample?

GabeSJones, 2019-03-11:
I'd like to base some code on the C# sample linked from this page, but it does not seem to include any licensing information.  Could you update it to include a license?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ead14b61-61e5-d45f-d5c9-d3d4c0bcf93d
* Version Independent ID: 31c51703-a968-0a6a-bba0-bdb47619b018
* Content: [Hardware dashboard API - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/dashboard-api#code-examples)
* Content Source: [windows-driver-docs-pr/dashboard/dashboard-api.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/dashboard-api.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-03-11:
Hi @GabeSJones.
Each of the C# sample foles contains the following statement:

// Sample Code is provided for the purpose of illustration only and is not intended to be used in a production environment. 
// THIS SAMPLE CODE AND ANY RELATED INFORMATION ARE PROVIDED "AS IS" WITHOUT WARRANTY OF ANY KIND, 
// EITHER EXPRESSED OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE IMPLIED WARRANTIES OF MERCHANTABILITY 
// AND/OR FITNESS FOR A PARTICULAR PURPOSE 

Thanks,
Eliot

compenguy, 2019-03-11:
This license amounts to a non-license.  It essentially says "you can't use this, write your own".

MIT would be just fine, and it's the license that's attached to the [hpc-api project](https://aka.ms/hpc_async_api_samples) linked from [the same page](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/dashboard-api#code-examples).

EliotSeattle, 2019-03-11:
Thanks, @compenguy. I'll look into it.

EliotSeattle, 2019-04-29:
Hi All.
The license states the following:

Permission is hereby granted, free of charge, to any person obtaining a copy
    of this software and associated documentation files (the "Software"), to deal
    in the Software without restriction, including without limitation the rights
    to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
    copies of the Software, and to permit persons to whom the Software is
    furnished to do so, subject to the following conditions:

    The above copyright notice and this permission notice shall be included in all
    copies or substantial portions of the Software.

    THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
    IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
    FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
    AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
    LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
    OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
    SOFTWARE.

You may use the code for any purpose outlined in the description, with attribution as stated, and with no warranty by Microsoft.

closed 2019-04-29 by EliotSeattle

GabeSJones, 2019-05-03:
Thank you for the clarification.  Is it possible to include this license in the source code archive at <http://download.microsoft.com/download/C/F/4/CF404E53-87A0-4204-BA13-A64B09A237C1/HardwareApiCSharpSample.zip>?

EliotSeattle, 2019-05-03:
Hi @GabeSJones.
I see that the .cs files in the downloadable sample have a more stringent license than the ones on GitHub. I don't have the authority to make a change to that, but I will reach out to the team that created the sample and see if something can't be done. I assume there were reasons for the different license, but I don't know what they are yet. I'll look into it.

Sorry that I'm not more help than that.

Eliot

EliotSeattle, 2019-05-08:
Hi @GabeSJones 
Well, I have another example of why I am not a lawyer. 
There is no *license* attached to the HardwareApiCSharpSample.zip files and the code in them. There is a *disclaimer* at the top of each file, though. You should look over the disclaimer and determine for yourself how comfortable you are using the sample, and for what purpose, but there is no particular license or agreement.
We are working, though, on how to make the terms more clear for others, and for that I thank you.
Hope this helps,
Eliot