MicrosoftDocs/windows-driver-docs #1396: Second sentence is incorrect.

FrankHeimes, 2019-03-12:
The second sentence states that only 'add' and 'delete' are allowed. But the code block beneath that sentence shows that 'query' (and /? for that matter) are allowed, too.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 561f2c34-2b98-1b84-a94a-8e580133a96e
* Version Independent ID: ae25edff-62f7-8659-40a1-4ff6f3e842f6
* Content: [SymStore Command-Line Options - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/symstore-command-line-options)
* Content Source: [windows-driver-docs-pr/debugger/symstore-command-line-options.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/symstore-command-line-options.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-12:
@FrankHeimes - Thanks for pointing out this error. A fix is on the way.

DOMARS, 2019-03-18:
@FrankHeimes - Fixes to second sentence are now live. Thanks for pointing out this issue.

closed 2019-03-18 by DOMARS