MicrosoftDocs/windows-driver-docs #1397: Windows internals 7 says command !pcr is deprecated

n4r1b, 2019-03-12:
In chapter 2 of Windows Internals 7 they mention !pcr extension is deprecated and shows incorrect data, while the documentation doesn't mention anything about it. 
Maybe it should be added as a remark.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f182853e-a14f-a8bf-b974-a280435c2676
* Version Independent ID: a46ba80b-a73a-c108-c48e-10a0a4758e48
* Content: [pcr - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-pcr#feedback)
* Content Source: [windows-driver-docs-pr/debugger/-pcr.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-pcr.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-12:
@n4r1b - I will take a look. It does seem to be present in current day WinDbg.

```
3: kd> !pcr
KPCR for Processor 3 at ffff98006ef40000:
    Major 1 Minor 1
	NtTib.ExceptionList: ffff98006ef53fb0
	    NtTib.StackBase: ffff98006ef52000
	   NtTib.StackLimit: 0000000000000000
	 NtTib.SubSystemTib: ffff98006ef40000
	      NtTib.Version: 000000006ef40180
	  NtTib.UserPointer: ffff98006ef40870
	      NtTib.SelfTib: 0000007bbaad3000

	            SelfPcr: 0000000000000000
	               Prcb: ffff98006ef40180
	               Irql: 0000000000000000
	                IRR: 0000000000000000
	                IDR: 0000000000000000
	      InterruptMode: 0000000000000000
	                IDT: 0000000000000000
	                GDT: 0000000000000000
	                TSS: 0000000000000000

	      CurrentThread: ffff98006ef50100
	         NextThread: 0000000000000000
	         IdleThread: ffff98006ef50100

	          DpcQueue: Unable to read nt!_KDPC_DATA.DpcListHead.Flink @ ffff98006ef42f80

3: kd> !thread
THREAD ffff98006ef50100  Cid 0000.0000  Teb: 0000000000000000 Win32Thread: 0000000000000000 RUNNING on processor 3
Not impersonating
DeviceMap                 ffff8500ae213240
Owning Process            fffff80605c089c0       Image:         Idle
Attached Process          ffffaf8a12ece080       Image:         System
Wait Start TickCount      0              Ticks: 66960069 (12:02:37:31.078)
Context Switch Count      3073255        IdealProcessor: 3             
UserTime                  00:00:00.000
KernelTime                03:27:19.218
Win32 Start Address nt!KiIdleLoop (0xfffff80605862e00)
Stack Init ffffec84a5645dd0 Current ffffec84a5645d60
Base ffffec84a5646000 Limit ffffec84a5640000 Call 0000000000000000
Priority 0 BasePriority 0 PriorityDecrement 0 IoPriority 0 PagePriority 0
Child-SP          RetAddr           : Args to Child                                                           : Call Site
ffffec84`a56458e8 fffff806`0a141404 : 00000000`00000000 ffffaf8a`1a7b30f0 ffff9800`6ef40180 ffffaf8a`1a7b3000 : 0xfffff806`0a14136f
ffffec84`a56458f0 00000000`00000000 : ffffaf8a`1a7b30f0 ffff9800`6ef40180 ffffaf8a`1a7b3000 00000000`00000046 : 0xfffff806`0a141404
```

n4r1b, 2019-03-13:
Yes, I also realized results seem to be correct, maybe they fixed since the release of the book. In that case the Issue may be closed. 

But I'm still a little bit skeptical because the book clearly states: 
[image]

DOMARS, 2019-03-13:
@n4r1b - Thanks for the clarification. I see the note on page 77 and will check with the dev team.

DOMARS, 2019-10-08:
@n4r1b - There is a work task to update the output for the !pcr command. Until that work is done I added a note indicating that this command may display incorrect output. When the work task is completed and the output is updated, I will remove this note. Thanks for pointing out this issue and for your patience with our response.

closed 2019-10-08 by DOMARS