MicrosoftDocs/windows-driver-docs #1398: x srv!SrvComputerName cannot work in Windows 10.

wayyt, 2019-03-12:
x srv!SrvComputerName cannot work in Windows 10. But I found a great article by SNoone to fix this issue. It introduces "r @$t0 = @@masm(mrxsmb!SmbCeContext); dx (nt!_UNICODE_STRING[4])(@$t0)" or "dpu mrxsmb!SmbCeContext" to get computer name.

The original article is here:
https://www.osr.com/blog/2018/06/22/finding-computer-name-crash-dump-2018-edition/
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2a09232d-fbb3-2766-29d7-a742f0bde3c0
* Version Independent ID: 5f34f241-dcf4-569c-fdfe-4647103ee379
* Content: [Extracting Information from a Dump File - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/extracting-information-from-a-dump-file)
* Content Source: [windows-driver-docs-pr/debugger/extracting-information-from-a-dump-file.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/extracting-information-from-a-dump-file.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-03-13:
@wayyt - Thanks for the pointer to Scott's article. Looks like this topic needs updating. 
In my limited testing, with Windows desktop 1809, I was able to display mrxsmb!SmbCeContext, but it did not contain any of the expected information - just empty strings, so wondering if this tip is still valid. Is there a pre-requisite step or specific type of dump that this is valid with? Do you need to be in a specific context?

Another technique here is to do a raw memory search, but in addition to determining the range, you need a valid target string.

DOMARS, 2020-12-18:
@wayyt - Thanks again for this feedback on validating / updating this topic and the pointer to the OSR article. I have created and internal work tracking item, so closing this issue. Good luck with your Windows debugging.

closed 2020-12-18 by DOMARS