MicrosoftDocs/windows-driver-docs #1400: NewInterfaceUsage

aplocher, 2019-03-12:
Can you please include information about what NewInterfaceUsage is exactly?

Also, what is SkipBOSDescriptorQuery and SkipContainerIdQuery and are they still valid?

Thank you

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 32122b55-0ed3-e2df-4bc4-f37c52d5af65
* Version Independent ID: d24d5892-5cb7-05aa-f54a-db06d4f3b7d9
* Content: [USB Device Registry Entries - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-device-specific-registry-settings#feedback)
* Content Source: [windows-driver-docs-pr/usbcon/usb-device-specific-registry-settings.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/usb-device-specific-registry-settings.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-03-13:
Hi @aplocher, thanks for the question. Let me see what I can find out about those for you.

duncanmacmichael, 2019-03-13:
@aplocher Could you be more specific about where you got these registry keys? I'm trying to ask the USB team about them but they need more information first.

desowin, 2019-05-29:
@duncanmacmichael I noticed `NewInterfaceUsage` under some of the keys listed in HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags

Example key under which the `NewInterfaceUsage` REG_DWORD appears is HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\046DC52F2200

I would like to know what it means as well as the original poster.

duncanmacmichael, 2019-05-31:
Thanks for adding on @desowin. I confirmed the presence of `NewInterfaceUsage` and `SkipContainerIdQuery` on my own PC, so I've refreshed the thread I had going with the USB team to see what I can find out.

duncanmacmichael, 2019-05-31:
Hi everyone, here's a summary of my findings on this issue.

These flags are among the many, many registry keys/values in the OS that are undocumented by design. They are not developer-facing knobs or settings, and developers and end-users are not meant to modify them. Thus, the better question is what problem are you trying to solve? If you are experiencing an issue during driver development, changing these keys/values is not likely the right way to solve whatever you're trying to do. 

Thanks for the question. If there are things missing from the docs that do need to be documented, I'm more than happy to do so. But, not all things are meant to be documented. If you're having a driver development issue that isn't directly related to the documentation, I'd suggest looking on either the [WDK forum](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk) or the [OSR forum](https://community.osr.com/), where you can find many other Windows driver developers and experts.

Thanks! Hope this helps.

closed 2019-05-31 by duncanmacmichael

desowin, 2019-05-31:
@duncanmacmichael I was thinking the NewInterfaceUsage somehow related to MS OS descriptors. I have noticed that HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\046DC52F2200\osvc value is 0x0000. However, the URB_FUNCTION_GET_MS_FEATURE_DESCRIPTOR is being issued to this device. I would like to know why?

duncanmacmichael, 2019-06-01:
@desowin Given that these flags are undocumented for internal use, I can't say exactly how or why the OS would issue that to a particular device. All osvc means is that the OS queried the device for a proprietary Microsoft-defined USB descriptor but it doesn't explain the reason. See [Microsoft OS Descriptors for USB Devices](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/microsoft-defined-usb-descriptors) for more info about the nature of those descriptors. 

Anectodally, various keys under _usbflags_ on my own PC sometimes have osvc, some don't, some have 0x0000 for its value, some have other values, some have other registry entries. There's no consistent pattern and it probably depends on the devices in your PC and their individual properties.